arendst/Sonoff-MQTT-OTA-Arduino #256: Feature Request - Reset to default from Wall Switch

jaysuk, 2017-01-20:
I am rolling installation of the sonoff's throughout my house.
This involves, in a couple of places, lifting floor boards etc. This means I will not have access to the sonoff if a bad upload occurs.
Could a number of button presses in quick succession (say 10) be implemented to allow the sonoff to be reset back to default settings?

arendst, 2017-01-21:
I'm looking into it but I'm afraid that it will be too late for the software to react on your ten presses when you feel the sonoff doesn't do what you expected (because it hangs or handles an exception loop).

I wouldn't nail the floor boards to tight...

jaysuk, 2017-01-21:
If it's just a hang, I would turn the mains off and on again.
I'm more thinking of if there's been a major change in the software, similar to the transition between version 2 and 3. Virtually all my sonoffs needed to be reset to default and then reprogrammed. That's the time my lights would be broken.