arendst/Sonoff-MQTT-OTA-Arduino #257: Sonoff Dual - Wall switch strange behaviour

nmrds, 2017-01-20:
I'm using one Sonoff Dual with 2 lamps and 2 wall push-buttons. I flashed the version 3.2.3 with this settings enabled in user_config.h
#define MODULE                 SONOFF   
#define SWITCH_MODE      PUSHBUTTON
#define USE_WALL_SWITCH

The 2 wall push-buttons are connected to button 0 and button 1 header.

If I use the webpage I can control the 2 lamps well individually, and the same with MQTT. 
Button 0 works great with lamp 0, but button 1 behaves strangely. When I push the button lamp 1 blink and toggles lamp 0 state.

Regards and thanks for all the work

arendst, 2017-01-21:
Using the button 0 and 1 header is not supported. What happens is that you control the relays directly without intervention from the esp8266 and my code but do send status information to the esp8266 and my code via the serial interface. My code reacts to the status messages and changes relays again leading to your observed strange results.

The settings in user_config.h are used when a switch is connected to a designated GPIO port, mostly pin 14.

mrjemson, 2017-01-25:
Is there a plan to support these headers in the near future?
I have a couple of Duals I wanted to use so I didn't have to install two TH10s inside the same wall cavity connected to one wall plate but still want to control independently.

hamwong, 2017-01-29:
HI, I am a newbie on this, do you mind share how to connect wall switch with sonoff? does it also works with sonoff basic?

enen, 2017-01-31:
I've realised that a doble click un the button 1 header toggle relay 2 indeed.

merlinmb, 2017-02-10:
Weird behaviour on the Dual - to which I have wire two n/o push buttons onto the headers:
The one pin combo (shorting to ground) turns on and off relay 1
 ..double clicking on this push botton toggles only relay 2, but with a on-->off-->on (maintain) 

The second pin combo (shorting to ground) turns on and off both relays