arendst/Sonoff-MQTT-OTA-Arduino #258: USE_WEMO_EMULATION activation over webinterface

thommycgn, 2017-01-21:
Hi! 

I'm always using your binary file from 
https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/blob/master/api/arduino/sonoff.ino.bin
to update the sonoffs. 

But  #define USE_WEMO_EMULATION    &&   #define USE_HUE_EMULATION     
are deactivated/uncomment by default. 

Would it be possible to make it accessible over the webinterface instead of the hardcoded version?

arendst, 2017-01-21:
I'm working on an integrated version where I would indeed allow enabling Alexa in realtime.

thommycgn, 2017-01-27:
Why do you not activate Wemo-Emulation by default till you make it accessible by web interface?

arendst, 2017-01-27:
Most people do not use Wemo or Hue emulation and when activated it really has a negative effect on overall performance especially when there is a lot of uPnP traffic in the air.

thommycgn, 2017-01-27:
Ok, I understand. I will keep waiting then ;)