arendst/Sonoff-MQTT-OTA-Arduino #261: Sonoff Touch two presses

Beetle-II, 2017-01-21:
Hello!
Not work two short presses on Sonoff Touch.
This is normal?

arendst, 2017-01-21:
It might not work as the sonoff touch button works a bit differently. I'll need time to investigate.

maizim, 2017-01-22:
I have the same issue and i will be very happy, if you could solve this problem.
Because my wife - and i can understand this - don´t like this button because of needing more then one attempt to switch the light on or of!
Thank you for this great projekt!!

mdelpire, 2017-01-22:
Hi,
Thanks a lot for this amazing job working perfectly. I confirm the fact that the button needs to be pressed twice in order to trigger the siwtch.

arendst, 2017-01-22:
Can you send the output of command ```status 0```?

mdelpire, 2017-01-22:
Here you are :)

15:46:46 HTTP: Handle console
15:46:46 MQTT: Receive topic cmnd/sonoffTouch/status, data size 1, data 0
15:46:46 MQTT: DataCb Topic sonoffTouch, Group 0, Index 1, Type STATUS, Data 0 (0)
15:46:46 MQTT: stat/sonoffTouch/RESULT = 3.1.1, 1, sonoffTouch, 0, POWER, 1, 2, 1, 1, 1, 1, 0, 0
15:46:46 MQTT: stat/sonoffTouch/RESULT = PRM: Baudrate 115200, GroupTopic sonoff2s, OtaUrl http://192.168.1.101:80/api/arduino/sonoff.ino.bin, Uptime 309 Hr, BootCount 18, SaveCount 258
15:46:46 MQTT: stat/sonoffTouch/RESULT = FWR: Program 3.1.1, Boot 4, SDK 1.5.3(aec24ac9)
15:46:46 MQTT: stat/sonoffTouch/RESULT = LOG: Seriallog 4, Weblog 4, Syslog 4, LogHost 192.168.1.101, SSId1 WiFi-2.4-950F, Password1 xxxxxxxxxxxxx, SSId2 indebuurt2, Password2 VnsqrtnrsddbrN, TelePeriod 300
15:46:46 MQTT: stat/sonoffTouch/RESULT = MEM: ProgramSize 385kB, Free 552kB (Heap 13kB), SpiffsStart 940kB (64kB), FlashSize 1024kB (1024kB)
15:46:46 MQTT: stat/sonoffTouch/RESULT = NET: Host sonoffTouch-3383, IP 192.168.1.2, Gateway 192.168.1.1, Subnetmask 255.255.255.0, Mac 60:01:94:82:ED:37, Webserver 2, WifiConfig 1
15:46:46 MQTT: stat/sonoffTouch/RESULT = MQT: Host 192.168.1.101, Port 1883, Client DVES_%06X (DVES_82ED37), User DVES_USER, Password DVES_PASS, MAX_PACKET_SIZE 400, KEEPALIVE 15
15:46:46 MQTT: stat/sonoffTouch/RESULT = TIM: UTC Sun Jan 22 14:46:46 2017, Local Sun Jan 22 15:46:46 2017, StartDST Sun Mar 26 02:00:00 2017, EndDST Sun Oct 29 03:00:00 2017

arendst, 2017-01-22:
I see no strange config errors. If I touch once it just switches as does tapping it twice too as expected:
```
16:29:18 APP: Multipress 1
16:29:18 MQTT: stat/touch1/RESULT = {"POWER":"ON"}
16:29:18 MQTT: stat/touch1/POWER = ON
16:29:18 Config: Saved configuration (1160 bytes) to flash at F7 and count 38
16:29:27 APP: Multipress 1
16:29:28 MQTT: stat/touch1/RESULT = {"POWER":"OFF"}
16:29:28 MQTT: stat/touch1/POWER = OFF
16:29:28 Config: Saved configuration (1160 bytes) to flash at F8 and count 39
16:29:29 APP: Multipress 1
16:29:30 APP: Multipress 2
16:29:30 MQTT: stat/touch1/RESULT = {"POWER":"ON"}
16:29:30 MQTT: stat/touch1/POWER = ON
16:29:30 Config: Saved configuration (1160 bytes) to flash at F7 and count 40
```
If you try to switch your touch how many APP: Multipress x do you see?

mdelpire, 2017-01-22:
Here is the log when I press twice the button to switch the light on : 

16:42:09 APP: Multipress 1
16:42:10 MQTT: domoticz/in = {"idx":11, "nvalue":1, "svalue":""}
16:42:10 MQTT: stat/sonoffTouch/POWER = ON
16:42:10 Config: Saved configuration (924 bytes) to flash at F7 and count 260
16:42:10 MQTT: Receive topic domoticz/out, data size 211, data {
   "Battery" : 255,
   "RSSI" : 12,
   "dtype" : "Light/Switch",
   "id" : "0001405B",
   "idx" : 11,
   "name" : "Entrée",
   "nvalue" : 1,
   "stype" : "Switch",
   "switchType" : 

What's the frequency when you press twice your button? do you "double click" or do you press twice with one seconde in between? I need to press it with one seconde in between.

arendst, 2017-01-22:
If I tap the button twice with more than 1 second interval it switches on at the first tap and off at the second tap. If I tap it quickly (within 1 second as that is the window I allow for multipress) then it receives two presses showing by multipress and switches the relay as expected as it executes the second tap option:
- 1 tap switch relay or executes ButtonTopic if it is NOT "0" (yours is "0" and OK)
- 2 taps switches relay directly

What happens if you tap it twice within 1 second?

mdelpire, 2017-01-22:
Well strange enough... now if I click once it works multipress1 and if I click twice quickly it's multipress 2... I don't get it... I'll try to check again later...

maizim, 2017-01-22:
I would only like to press the button once and the light should go on or off.
I thought yes it is s.Sensor, but I simply do not find regularity. Sometimes the light goes with a touch and then I need again several attempts to switch the light. It is very strange.

arendst, 2017-01-22:
Stop logging too many on too many media. I guess your touch is not connected to serial anymore so set seriallog to 0. Having syslog set to 4 with domoticz flooding MQTT  is a path to disaster too. set it to 2 at most. When done debugging also set weblog to 2.

Now you should be able to enjoy your sonoff touch with just a single tap  (I hope...)

mdelpire, 2017-01-23:
Hi, 
I stoped all the logs (set to 0 for serial, syslog and web) but still the same behaviour... 90% of the time, two presses are needed.

maizim, 2017-01-23:
For me it is the same.
It seems like the touchfunction is a little bit more sensitiv, but i often need more attemps to switch the light.
Can I reduce traffic, if i disable other functions (e.g. mqtt)?

maizim, 2017-02-02:
Today, i have tested another thin script and it runs smooth and every touch is successfull.
Then i have cross tested the actual version 3.2.13 and i have the same problems with the touch.
I think, this super script is to big for the touch (and the prozessor). 
Is there a possility to make it smaller? I don´t need the whole support of this firmware for the touch. I think a web-interface and the possibility to control the device with http: requests (e.g. http://192.168.1.232/cm?cmnd=power%20on) is perfect for me and it could be working nice and smooth without probles.
Thanx for your great work!

arendst, 2017-02-02:
Try https://github.com/arendst/Sonoff-Tasmota and disable MQTT and Wemo and Hue in the web interface or just don't compile them in.

Same for this version though, just disable MQTT in user_config.h

maizim, 2017-02-02:
Now, I am back on 3.1.19, because the higher version "kills" the sonoff touch".
After some the device is suddenly unreachable.
I can only reactivate the sonoff touch by flashing it again!
My first question is, where in the user_config.h can I disable the MQTT?
And second can I flash it ota, after exporting it to a bin-file?
thanx ;-)

dtavares07, 2017-03-29:
Can you guys tell me in witch gpio the touch button is connected to? 

Best regards,
Daniel Tavares