arendst/Sonoff-MQTT-OTA-Arduino #262: Status Meaning - 8th field - 1 vs. 0

jaquesp, 2017-01-21:
Hi there - I have two sonoff devices, one of which doesn't retain power status at startup. In all ways the configuration of the devices is the same. However, if I check the status of the two devices they are slightly different:

```
<MqttMessage id="8966" timestamp="1485004167242" topic="cmnd/lrstring/status"></MqttMessage>
<MqttMessage id="8967" timestamp="1485004167254" topic="stat/lrstring/RESULT">3.1.16, 1, lrstring, lrstring, POWER, 1, 3, 0, 1, 1, 1, 1, 1</MqttMessage>
<MqttMessage id="8968" timestamp="1485004174097" topic="cmnd/lilystring/status"></MqttMessage>
<MqttMessage id="8969" timestamp="1485004174113" topic="stat/lilystring/RESULT">3.1.16, 1, lilystring, lilystring, POWER, 0, 3, 1, 1, 1, 1, 1, 1</MqttMessage>
```

Apart from the obvious POWER status difference (the first device "lrstring" was on) - what does the 8th value represent? 0 vs. 1?

Thanks in advance....

arendst, 2017-01-21:
Try with messageformat 1 which should explain each field.

jaquesp, 2017-01-21:
I  have now tried that but a subsequent call to status returns nothing - am I missing something?

```
<MqttMessage id="9530" timestamp="1485025587886" topic="cmnd/lilystring/messageformat">1</MqttMessage>
<MqttMessage id="9531" timestamp="1485025587893" topic="stat/lilystring/RESULT">{"MessageFormat":"JSON"}</MqttMessage>
<MqttMessage id="9532" timestamp="1485025590454" topic="cmnd/lilystring/status"></MqttMessage>
```

Returns nothing....

Should I see something?

arendst, 2017-01-21:
When you now execute command status you should see

20:18:35 MQTT: stat/pow1/RESULT = {"Status":{"Module":5, "FriendlyName":"Pow1", "Topic":"pow1", "ButtonTopic":"0", "Subtopic":"POWER", "Power":0, "PowerOnState":3, "LedState":0, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}

jaquesp, 2017-01-21:
I don't see anything. It never returns a status.. Do I have an old firmware version? 3.1.16? 

Thanks for your help.

arendst, 2017-01-21:
Execute this: ```mosquitto_sub -t stat/# -t tele/# -t cmnd/# -v```

Then perform a ```status``` command which should show this:
```
stat/pow1/RESULT {"Status":{"Module":5, "FriendlyName":"Pow1", "Topic":"pow1", "ButtonTopic":"0", "Subtopic":"POWER", "Power":1, "PowerOnState":3, "LedState":0, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
```

jaquesp, 2017-01-21:
Thanks again for getting back to me:

```
pi@raspberrypi:~ $ mosquitto_sub -u password -P U3sjB4MtVLvH -t stat/# -t tele/# -t cmnd/# -v
stat/lilystring/POWER ON
tele/lilystring/LWT Online
cmnd/lilystring/POWER ON
```

Then I processed this:

```
mosquitto_pub -u evsefgcx -P password -t cmnd/lilystring/messageformat -m "1"

cmnd/lilystring/messageformat 1
stat/lilystring/RESULT {"MessageFormat":"JSON"}
```

And then this:

```
mosquitto_pub -u evsefgcx -P password -t cmnd/lilystring/status -m ""
cmnd/lilystring/status (null)
```

With no response. If I process a status2 command, I get the following:

```
mosquitto_pub -u evsefgcx -P password -t cmnd/lilystring/status -m "2"
cmnd/lilystring/status 2
stat/lilystring/RESULT {"StatusFWR":{"Program":"3.1.16", "Boot":4, "SDK":"1.5.3(aec24ac9)"}}
```

Nothing else works with any of meh devices flashed with 3.1.16.

Thanks again.

arendst, 2017-01-21:
What happens with a ```status 0``` command?

jaquesp, 2017-01-21:
Same result as with a `status 2` command:

```
mosquitto_pub -u evsefgcx -P U3sjB4MtVLvH -t cmnd/lilystring/status -m "0"

cmnd/lilystring/status 0
stat/lilystring/RESULT {"StatusFWR":{"Program":"3.1.16", "Boot":4, "SDK":"1.5.3(aec24ac9)"}}
```

arendst, 2017-01-21:
Ah. Did you change MQTT_MAX_PACKET_SIZE in pubsubclient.h as signaled in sonoff.ino:
```
 * ====================================================
 * Prerequisites:
 *   - Change libraries/PubSubClient/src/PubSubClient.h
 *       #define MQTT_MAX_PACKET_SIZE 400
 *
 *   - Select IDE Tools - Flash size: "1M (64K SPIFFS)"
 * ====================================================
```

jaquesp, 2017-01-21:
Ah.... No! My omission! Let me change that, reflash and try again! Thanks for all your help!

closed 2017-01-24 by arendst