arendst/Sonoff-MQTT-OTA-Arduino #264: Pow restarts after update to 3.2.6 ... ?

ecsfang, 2017-01-22:
Hi Theo!

I updated my Pow yesterday to 3.2.6 (from 3.1.2 I think it was), and since then it just restarts within 5-10 minutes ...
I noticed it because the TODAY_ENERGY was reset every now and then and not incremented as it has been doing.

I thought it had to do with mDNS which was enabled, but turning it off didn't change the behavior.

[image]

The consol log doesn't show anything, except that the log is restarted:

00:00:00 APP: Project sonoff DVES_20DCC1 (Topic annexet, Fallback DVES_20DCC1, GroupTopic pows) Version 3.2.6
00:00:00 Wifi: Connecting to AP2 TIGERGATE in mode 11N as annexet-7361...
00:00:04 Wifi: Connected
00:00:04 HTTP: Webserver active on annexet-7361 with IP address 192.168.1.225
13:50:38 MQTT: Attempting connection...
13:50:38 MQTT: Connected
13:50:38 MQTT: tele/annexet/LWT = Online (retained)
13:50:38 MQTT: tele/annexet/INFO = Sonoff Pow Module version 3.2.6, FallbackTopic DVES_20DCC1, GroupTopic pows
13:50:38 MQTT: tele/annexet/INFO = Webserver active for Admin on annexet-7361 with IP address 192.168.1.225
13:50:39 MQTT: stat/annexet/POWER = OFF
13:50:47 MQTT: tele/annexet/POWER = OFF
13:50:47 MQTT: tele/annexet/UPTIME = 0
13:50:47 MQTT: tele/annexet/AP = 2
13:50:47 MQTT: tele/annexet/SSID = TIGERGATE
13:50:47 MQTT: tele/annexet/RSSI = 84
13:50:47 MQTT: tele/annexet/YESTERDAY_ENERGY = 5.169
13:50:47 MQTT: tele/annexet/TODAY_ENERGY = 0.000
13:50:47 MQTT: tele/annexet/PERIOD_ENERGY = 0
13:50:47 MQTT: tele/annexet/CURRENT_POWER = 0
13:50:47 MQTT: tele/annexet/POWER_FACTOR = 0.00
13:50:47 MQTT: tele/annexet/VOLTAGE = 0
13:50:47 MQTT: tele/annexet/CURRENT = 0.000
13:50:47 MQTT: tele/annexet/TIME = 2017-01-22T13:50:47
13:53:23 MQTT: stat/annexet/POWER = OFF

Any ideas about what might has caused this?

Best regards,
Thomas

arendst, 2017-01-22:
Thomas,

I noticed almost the same and I think the problem is having changed the interrupt handlers from RAM to ROM due to the WS2812 code.

I suggest you change the following lines in file xsns_hlw8012.ino from:
```
Ticker tickerHLW;

//void hlw_cf_interrupt() ICACHE_RAM_ATTR;
//void hlw_cf1_interrupt() ICACHE_RAM_ATTR;

void hlw_cf_interrupt()  // Service Power
```
to:
```
Ticker tickerHLW;

void hlw_cf_interrupt() ICACHE_RAM_ATTR;
void hlw_cf1_interrupt() ICACHE_RAM_ATTR;

void hlw_cf_interrupt()  // Service Power
```
and let know if it helps. Enabling RAM again makes the WS2812 DMA code (using GPIO03 only) unable to compile but I'm not fond of the code anyway...

ecsfang, 2017-01-22:
Thanks for quick response!

I have recompiled and uploaded to the pow, so now I just have to wait a while and see ...
I will update the issue with the result!

Cheers!
Thomas

ecsfang, 2017-01-22:
Hi again!

Yes, can confirm that it works fine on my side now! :)

[image]

Dank u wel!

closed 2017-01-22 by arendst