arendst/Sonoff-MQTT-OTA-Arduino #265: Issue while compileing

ItsFletch, 2017-01-22:
Arduino: 1.6.10 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"

C:\Users\Danny\Desktop\Sonoff-MQTT-OTA-Arduino-master\sonoff\sonoff.ino: In function 'void setup()':

sonoff:2634: error: 'class HardwareSerial' has no member named 'baudRate'

   if (Serial.baudRate() != %d) {

              ^

sonoff:2634: error: expected primary-expression before '%' token

   if (Serial.baudRate() != %d) {

                            ^

sonoff:2634: error: 'd' was not declared in this scope

   if (Serial.baudRate() != %d) {

                             ^

C:\Users\Danny\Desktop\Sonoff-MQTT-OTA-Arduino-master\sonoff\webserver.ino: In function 'void handleInfo()':

webserver:1180: error: 'class EspClass' has no member named 'getCoreVersion'

   page += F("<tr><td><b>Core/SDK version</b></td><td>"); page += ESP.getCoreVersion(); page += F("/"); page += String(ESP.getSdkVersion()); page += F("</td></tr>");

                                                                      ^

exit status 1
'class HardwareSerial' has no member named 'baudRate'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

arendst, 2017-01-22:
The answer is right under your nose.

Hints: READ THE F*C*I*G WIKI. GO OVER THE ISSUES AS AT LEAST A CURRENTLY OPEN ONE CONTAINS THE ANSWER.

I'm tired...

ItsFletch, 2017-01-22:
Can you link it? Im sorry but if i would have seen it i would not have posted. Just wanting a little help....

arendst, 2017-01-23:
That's strange. Try re-installing the ESP8266 board library which might solve the issue.

ItsFletch, 2017-01-23:
I have countless times? Any other ideas on what could be causing it? Thanks

ItsFletch, 2017-01-23:
Getting this error now and it highlights the last line in xdrv_wemohue, The line reads #endif  // USE_WEMO_EMULATION || USE_HUE_EMULATION

Error messege: 

C:\Users\Danny\Desktop\Sonoff-MQTT-OTA-Arduino-master\sonoff\xdrv_wemohue.ino: In function 'void setup()':

xdrv_wemohue:2638: error: 'class HardwareSerial' has no member named 'baudrate'

xdrv_wemohue:2638: error: 'baudrate' was not declared in this scope

exit status 1
'class HardwareSerial' has no member named 'baudrate'

ItsFletch, 2017-01-23:
Solved it, i just copied my config file, downloaded again put my config file in and it compiled, Thanks for the help anyways!

closed 2017-01-24 by arendst