arendst/Sonoff-MQTT-OTA-Arduino #266: Issue when toggling from Web Interface

jampola, 2017-01-23:
Replicated the issue running Debian Jessie 64bit in the following browsers:

Firefox (45.6.0 ESR) 
Chromium (53.0.2785.143) 
Chrome (55.0.2883.87)

Firefox console upon POST:

> A form was submitted in the windows-1252 encoding which cannot encode all Unicode characters, so user input may get corrupted. To avoid this problem, the page should be changed so that the form is submitted in the UTF-8 encoding either by changing the encoding of the page itself to UTF-8 or by specifying accept-charset=utf-8 on the form element.

> The character encoding of the HTML document was not declared. The document will render with garbled text in some browser configurations if the document contains characters from outside the US-ASCII range. The character encoding of the page must be declared in the document or in the transfer protocol.

However I have no issue with curl http://sonoffip/?={1,0} or calling that URL directly in the address bar.

I have recompiled with:

` "<meta http-equiv=\"Content-type\" content=\"text/html;charset=UTF-8\">"`

This removes the browser complaining about char encoding but the issue still persists.

arendst, 2017-01-23:
I can only test on windows and there I see no problems regarding toggling from the webpage with either firefox 50.1.0, IE 11, Edge or Chrome 55.0.2883.87

What does fail is firmware uploading using Firefox while the others work just as well.

I did add your utf-8 suggestion in future versions.

jampola, 2017-01-23:
Okay, so I just tried it on Chome on Android and IE10 on Windows 7, same issue. I'll try do a fresh reflash tomorrow on another Sonoff Basic. And sorry, neglected to mention I was flashing 3.2.7, which appears to be the latest.

> I did add your utf-8 suggestion in future versions.

Great! Not that I see it making a huge difference apart from some pesky console logs complaining about UTF-8 :)

arendst, 2017-01-24:
See #232

closed 2017-01-24 by arendst