arendst/Sonoff-MQTT-OTA-Arduino #267: Feature Request: Add support for NodeMCU.

gizmochief7, 2017-01-23:
I was hoping to reuse your library for testing on a NodeMCU and just connect up relays, leds, and/or reset buttons as needed depending on what type of Sonoff device i was trying to simulate. Your code compiled and pushed to the NodeMCU without any errors. I was able to hit the web server and toggle on and off the device with feedback in both the logs and the led status. However, I was not able to get any of the relay pins to actually trigger nor get the hardware button presses to fire an event to the logs or led.

arendst, 2017-01-23:
If you execute command ```model 0``` things might change for the good as the software currently probably thinks it's a dual.

I suggest you use the next_gen version as there you have more control over the specific modules supported and could even update the sonoff_template.h file with your specific hardware configuration.

jesusvallejo, 2017-01-24:
Ive tried the code on an nodemcu v1 and works just fine

closed 2017-01-24 by arendst

gizmochief7, 2017-01-25:
Yep, everything now works fully after running the model 1 command. You were right that it was thinking it was a Sonoff Dual at first. Thanks for the help and keep up the great work.