arendst/Sonoff-MQTT-OTA-Arduino #271: Don't go to Wifi AP mode when can't connect to Wifi

Imi3est, 2017-01-24:
Hi,

I am testing the sonoff device and I found a problem with it. When it can't connect to wifi network it always goes to Access Point mode. I tested like this:
 - Sonoff connected to wifi (MQTT messages arrived)
 - I renamed wifi ssid
 - Sonoff went to AP mode
This is bad because my neighbours can change its's settings. It is enough if it goes to AP mode when I press the button four times on sonoff.  At webserver I don't want to disable admin mode because if I change something in evironment I don't want to reflash sonoff to change wifi or mqtt parameters.
I  have this setting:  #define WIFI_CONFIG_TOOL       WIFI_RESTART
I also tried WIFI_RETRY, but AP still appears when can't connect.

Is it a real problem or I do something wrong?
Thank you!!

arendst, 2017-01-24:
Try wificonfig 4.

Imi3est, 2017-01-24:
Thank you! wificonfig 4 command solved the problem. I thought the configuration of sonoff should done by changing userconfig.h but now I discowered that there are command for that.

Thanks for fast answer!!

closed 2017-01-24 by Imi3est