arendst/Sonoff-MQTT-OTA-Arduino #272: Consider power save option ?

marcelveldt, 2017-01-24:
First of all thank you for your amazing job on this! Bedankt!

Before I discovered your work I was experimenting with the sonoffs myself.
I measured the power consumption of the sonoff device with a simple Voltcraft (conrad) power monitor  and concluded that by default, the sonoff (and any other ESP module) will consume about 0.6/0.7 of watts continuously at 230V

It makes no difference if I use my own simple sketch or your complete package.

This power consumption can be greatly reduced with 2 small code changes:

1) Set the sleep mode to light sleep: 
wifi_set_sleep_type(LIGHT_SLEEP_T);

2) Add a delay in the loop execution. E.g. 250ms or 500ms

With the delay set to 500ms the power consumption will drop down to 0.2watts. With a lot of sonoff devices this can make a huge difference. The only downside is that commands will respond a little bit later but hardly noticeable with only 0,5 seconds of delay in execution.

Could you consider adding this as an optional setting ? If enabled, the device has energy savings enabled. Or maybe even enable it during a specific time slot or something ?

I just tried to add it to your code as a simple test but ran into the issue that you have all kind of counters in place that will trip because of the large delay inside the main loop.

Thanks in advance for considering !

arendst, 2017-01-26:
Implemented in version 3.2.12 just released as command sleep 0 - 250 mSec.

Expect problems with Sonoff Pow (interrupt driven) and button/key/switch missing activation with sleep times larger than 10 mSec. 

Marcel, is there any advantage of having 10 mSec sleep as opposed to 0?

marcelveldt, 2017-01-26:
Thanks, I'll go test it right away!

In my own testcode I had best results with a delay of around 200ms (or higher but that's messy with the buttons). In that case the power usage drops down to 0.2 watts measured with a pretty accurate meter on the 230V input.

With the delay set to 10 or 20ms the power consumption drops to around 0.4ms

I'll test the new code and report back.

marcelveldt, 2017-01-26:
OK, test results...

I quickly ported your new code to the next_gen version as that's what I currently use for my sonoffs.
Also added a field for the config in the webconfig for easy testing.

Setting it to a 10ms delay is giving great results, the button is still very responsive and the power usage is at about 0.35 watts.

Setting it to a 100+ ms delay gives best power saving (0.25 watts) but requires to hold the button just a little but longer. Tried it btw with the sonoff touch as that is most sensitive with the button.

Overall great work!

Do you want me to PR the fix for the next_gen version or wil you add it later ?

arendst, 2017-01-26:
Thnx for testing. 

No need for next gen update as I already merged it.

marcelveldt, 2017-01-26:
OK great, thanks!

closed 2017-01-26 by marcelveldt