arendst/Sonoff-MQTT-OTA-Arduino #274: Alexa discovery issue

Ryanmaturi, 2017-01-24:
Cannot get Alexa to consistently discover WeMo or hue.  It has worked at times other times not.  have kakopappa git hub project (kakopappa/arduino-esp8266-alexa-multiple- wemo-switch) on another esp8266 module. Alexa discovery works every time.

thommycgn, 2017-01-27:
I have the same problem. Sometimes it's help to switch off the power and switch it on again. Sometimes you have to search more time for the devices and suddenly it's there.

Ryanmaturi, 2017-01-28:
Suspected my network could be the cause, so far this has not proven to be true. Tried another router (openwrt with miniupnp running). I cannot discover at all right now, the Alexa feature is more desirable to me than the MQTT. So I hope I can figure it out. I do plan on donating to the developer for all his hard work.  It's just strange that others have no problems.

btsimonh, 2017-01-30:
I just had Alexa unable to discover my devices.  Turned out that I had changed my MQTT server's host name, and when the Sonoff cannot connect to MQTT, it seems it is also unable to be discovered by Alexa....

Ryanmaturi, 2017-02-01:
On my network config I could not get it to discover at all. This consists of a old laptop with dual ethernet running pfsense (upnp services enabled), a ubiquity uap ac-lr wifi AP, a gigabit switch for the LAN ports, this setup has served me well for years.  Only when I switched out the pfsense laptop for a apple airport router to do the routing did I get Alexa to discover the sonoffs. And even that is hit or miss as described by others.  I do have another esp8266 running the WeMo emulator from kakopappa (another git hub project).  This unit has never failed to be discovered. Just curious what the difference may be. I noticed that when my sonoffs ARE discovered, their console states that they are using port 50000. That sounds a little high. I thought port 5000 was the traditional upnp multicast port?

Ryanmaturi, 2017-02-01:
I should add that I have had MQTT working great and sonoff discovery did not work, and was hit or miss with the apple router

crwhite57, 2021-06-27:
I had this problem too, but I figured out what was the problem and now I don't have that problem anymore. The sketches have this in them:

void loop()
{
  espalexa.loop();
  delay(1);
}

The delay is the problem, it make the loop go so fast, Alexa has no time or very little time to recognize the devices. This little change fixed everything for me:

void loop()
{
  espalexa.loop();
  delay(10);
}