arendst/Sonoff-MQTT-OTA-Arduino #275: Slampher (light fitting) support - half brightness in 'OFF'.

btsimonh, 2017-01-24:
Has anyone tried the firmware on a Slampher light fitting?  Although mentioned in the readme.md, I can't find any reference in the code.  The schematics seem pretty similar to the std. sonoff, apart from the 'b utton.
When programmed, the fitting shows half brightness in 'OFF' state, and full brightness in 'ON' state (incandescent bulb).
I'm thinking maybe the 'switch' input (GPIO0) is being toggled constantly by the hardware; but before I break out the 'scope, any advice?  Maybe I just got a faulty unit....  So if you've had one work, please comment....

btsimonh, 2017-02-15:
just installed:
Program version	3.9.16
Build Date/Time	Feb 14 2017/22:40:26
from http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin
where slampher is a module option (and is selected).
The behaviour is still the same; so maybe we have two different hardwares...

smadds, 2017-02-16:
Is that using the ESP8285 chip? If so you could try http://sonoff.maddox.co.uk/tasmota/sonoff-8285.ino.bin although Theo has mentioned that a separate compile should not be needed any more.

arendst, 2017-02-16:
It's using the ESP8266 and uses a Triac to switch the light. The combination of Triac and lightbulb might be the problem. Try to see what happens when you use an 'ancient' light bulb (_the ones we used to have before the power saving mafia took on and invented lights that are much more expensive, die just as quickly as the old ones and are uncontrollable_) instead of a led based lamp.

btsimonh, 2017-02-16:
I've tried two incandescent (old style :) ) bulbs... only 40W thought.  Edison screw is not common in the UK...
I'm coming to think I may have to risk my isolation transformer and 'scope.
I tried setting the output pin to 'OUTPUT_OPEN_DRAIN' rather than 'OUTPUT' with no good effect.  I could not find any other output pin controls that may affect it.
One thing I did find when searching google was that people got very different wifi performance from the ESP8266 depending upon if they pulled inputs down or not...  Just maybe if I set all unused pins to outputs, then the board would be electrically quieter.  It 'feels' like spurious triggering of the Triac; but I assume it worked with their firmware (but iTead refuse to release their firmware for me to re-program and test :( ).
Any ideas about how to obtain original firmware appreciated! (there must be an 'upgrade' http source, as the devices can natively upgrade their software).....

arendst, 2017-02-16:
I suppose you have the schematics of the Slampher available from the ITEAD website (https://www.itead.cc/wiki/Slampher) down the bottom.