arendst/Sonoff-MQTT-OTA-Arduino #276: *Question* - Understanding Switch Modes

mrjemson, 2017-01-25:
I can't find any documentation on the different Switch Modes and would like to get an understanding of what they each mean.

TOGGLE, FOLLOW, FOLLOW_INV, PUSHBUTTON or PUSHBUTTON_INV

I would have thought Toggle would allow you to use a normal latching wall switch and would toggle based on state change, but that doesn't seem to be the case in my testing.
As such I would like to get an understanding of what each mode *should* do, so I can correctly raise an issue if it is in face misbehaving.

arendst, 2017-01-25:
See Wiki - Commands all the way down...

mrjemson, 2017-01-25:
Thanks. I have it set to toggle, but it seems to be behaving like push button.
I will do some further testing to confirm and come back.

closed 2017-01-27 by arendst