arendst/Sonoff-MQTT-OTA-Arduino #277: RGB Support crashes the uC

altelch, 2017-01-25:
Hi,

sometimes switching to SCHEME 1 crashes the uC.  I also get crashes after about 35minutes running even if doing nothing the uC crashes with Exception (28). I tried DMA mode and Bitbanging mode, both crash after a while without interaction or running animation.

Here 2 times a SCHEME = 1 crash. 
00:00:50 MQTT: stat/sonoff/SCHEME = 1

Exception (0):
epc1=0x40107189 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff1c90 end: 3fff20a0 offset: 01a0

>>>stack>>>
3fff1e30:  2d31302d 30543130 30303a30 0030353a  
3fff1e40:  4000050c 00020000 60000b10 00000000  
3fff1e50:  00000002 3ffefea8 00000000 fffffffe  
3fff1e60:  ffffffff 3fffc6fc 00000001 3fff05c0  
3fff1e70:  3fffdad0 00000000 3fff06a4 40209b40  
3fff1e80:  00000001 00000000 00000002 40101ee2  
3fff1e90:  401051a4 3ffedde8 3ffee660 00000000  
3fff1ea0:  00000001 401043b3 3ffee690 00000b68  
3fff1eb0:  00000005 00000000 00000020 40101ee2  
3fff1ec0:  3ffea515 401051bf 3ffedf78 00000030  
3fff1ed0:  40102bf5 3ffedf78 7fffffff 00000000  
3fff1ee0:  4010743e 030db04f 3ffee9c8 40102d84  
3fff1ef0:  4010743e 00000000 00000000 4000050c  
3fff1f00:  4010743e 00000030 00000010 ffffffff  
3fff1f10:  4010743e 3fff1080 00000000 4000050c  
3fff1f20:  4010787f 00000030 00000010 4000050c  
3fff1f30:  4010743e 00000030 00000010 ffffffff  
3fff1f40:  40107480 00000032 000fa000 4000050c  
3fff1f50:  40000f83 00000030 00000010 ffffffff  
3fff1f60:  40000ea3 00000023 00000001 00000001  
3fff1f70:  3fffdab0 00000000 3fffd9d0 3fff1070  
3fff1f80:  00000000 00000000 00000001 3fff1068  
3fff1f90:  00000000 3fffdad0 3fff1070 00000030  
3fff1fa0:  00000000 3fffdad0 3fff1070 00000030  
3fff1fb0:  00000000 3fffdad0 3fff1070 00000030  
3fff1fc0:  00000000 3fffdad0 3fff1070 00000030  
3fff1fd0:  ffffffff 3fffc6fc 00000001 3fff1080  
3fff1fe0:  00000000 3fffdad0 3fff1070 00000030  
3fff1ff0:  3fffdad0 3fff1070 402175c4 3fff1080  
3fff2000:  3fff1fd8 00000000 3fff4974 4020fbae  
3fff2010:  3fff2040 00000001 00000012 4021767c  
3fff2020:  00000012 0000c7b6 3fff30e4 4020feaf  
3fff2030:  3fffdad0 00000000 3ffeffe0 4020e08d  
3fff2040:  3ffe92a0 00000000 000003e8 00000000  
3fff2050:  00000000 00000000 402175c4 3fff1080  
3fff2060:  3fffdad0 00000000 3fff1068 3fff1070  
3fff2070:  3fffdad0 00000000 3fff1068 4020e15a  
3fff2080:  3fffdad0 00000000 3fff1068 40217610  
3fff2090:  feefeffe feefeffe 3fff1080 40100718  
<<<stack<<<

00:05:26 MQTT: stat/sonoff/SCHEME = 1
00:05:26 Config: Saved configuration (1108 bytes) to flash at 3F7 and count 248
00:05:27 UPnP: Multicast disabled
00:05:27 MQTT: Attempting connection...

Exception (28):
epc1=0x40217e98 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3fff1a70 end: 3fff20b0 offset: 01a0

>>>stack>>>
3fff1c10:  4021430f 00000000 3fff0e90 40214317  
3fff1c20:  74716d5f 303a0074 37323a35 402142b2  
3fff1c30:  3fff1c0c 000004ba 3fff30a7 61636f6c  
3fff1c40:  745f006c 00007063 00000003 00000005  
3fff1c50:  00000004 00000005 005f0000 40205f00  
3fff1c60:  3ffe8ad5 7441203a 706d6574 676e6974  
3fff1c70:  6e6f6320 7463656e 2e6e6f69 3fff05cc  
3fff1c80:  3ffe852c 3fff0e90 00000000 40205c59  
3fff1c90:  4022deda 00000000 00000003 00000002  
3fff1ca0:  00000010 3fff1df0 00000000 3fff1d40  
3fff1cb0:  3fff0f98 00000160 00000160 4010020c  
3fff1cc0:  3fff0fc0 00000050 3fff34dc 4010068c  
3fff1cd0:  3fff0f98 000004b4 000004b4 4010020c  
3fff1ce0:  3fff0f98 000000e8 000000e8 4010020c  
3fff1cf0:  3ffe8524 00000000 3fff311c 4010068c  
3fff1d00:  76615320 63206465 69666e6f 40217468  
3fff1d10:  6e6f6974 31312820 62203830 4020fd22  
3fff1d20:  00000000 3ffeffe1 3ffeffec 4020fd49  
3fff1d30:  4000050c 3fff1d40 3ffeffec 4020fec8  
3fff1d40:  40203758 00000030 3ffeff98 3fff05cc  
3fff1d50:  3ffe852c 3fff0004 3fff06b0 40208c7c  
3fff1d60:  521d6b23 00000000 00000000 60000308  
3fff1d70:  00000020 00000040 00000063 3fff2a84  
3fff1d80:  00000031 0000001e 3ffefec4 00000030  
3fff1d90:  00000100 40004b14 00000400 3fff05cc  
3fff1da0:  3ffe8ea5 3ffe8ad5 00000002 40214503  
3fff1db0:  3fff0e90 00000050 3fff0e60 00000400  
3fff1dc0:  b4a944c0 00ffffff fea944c0 3fff05cc  
3fff1dd0:  3ffe8524 3fff0e60 00000002 402096db  
3fff1de0:  3ffe9a48 b4a944c0 3fff06b0 40205adb  
3fff1df0:  666e6f43 203a6769 65766153 6f632064  
3fff1e00:  6769666e 74617275 206e6f69 30313128  
3fff1e10:  79622038 29736574 206f7420 73616c66  
3fff1e20:  4174cccd 37463320 646e6120 756f6320  
3fff1e30:  3220746e 0000001e 00000031 3fff05cc  
3fff1e40:  3fffdad0 3ffeffe9 3fff09b0 402041a5  
3fff1e50:  4022f5ee 3fff06b0 3ffeffe1 3ffe8524  
3fff1e60:  4023043b b4a944c0 00000006 31206174  
3fff1e70:  3fff2ac4 4020ea5c 00000001 3fff05cc  
3fff1e80:  3ffe8524 3ffeffe1 3fff06b0 40209c51  
3fff1e90:  00000005 00000000 00000020 40102006  
3fff1ea0:  3ffea525 401052e3 3ffedf88 40102006  
3fff1eb0:  3ffee6a0 40101b9a 00000001 00000000  
3fff1ec0:  00000000 40101967 00000000 3ffee670  
3fff1ed0:  00000001 00000000 00000002 40102006  
3fff1ee0:  401052c8 3ffeddf8 3ffee670 00000100  
3fff1ef0:  00000000 00000000 0000001f 40105af5  
3fff1f00:  4000050c 3ffeadac 3ffeadac 00000001  
3fff1f10:  40106424 00000030 0000000d ffffffff  
3fff1f20:  402176a2 00000001 00000001 00000000  
3fff1f30:  00000000 00000000 00000000 fffffffe  
3fff1f40:  00000000 00000000 0000001f 40105af5  
3fff1f50:  4000050c 3fffdad0 3fff107c 00000030  
3fff1f60:  00000000 00000000 0000001f 40105af5  
3fff1f70:  4000050c 00000023 00000001 00000001  
3fff1f80:  40201e4f 00000030 0000000d ffffffff  
3fff1f90:  40215bdb 00008000 6000001c 00000001  
3fff1fa0:  00000003 00000190 00000000 fffffffe  
3fff1fb0:  ffffffff 3fffc6fc 00000001 3fff0fc0  
3fff1fc0:  00000000 3fffdad0 3fff107c 00000030  
3fff1fd0:  00000000 3fffdad0 3fff107c 00000030  
3fff1fe0:  00000000 3fffdad0 00000000 402113a4  
3fff1ff0:  3fff1d30 00000001 3fff1090 0004feaa  
3fff2000:  3fffdad0 3fff107c 402175f8 3fff1090  
3fff2010:  3fff1fe8 00000000 3fff3d44 4020f8a6  
3fff2020:  3fff2050 00000001 00000012 402176b0  
3fff2030:  00000012 0004fbe7 3fff311c 4020fba7  
3fff2040:  3fffdad0 00000000 3ffeffec 4020dd85  
3fff2050:  3ffe9298 00000000 000003e8 00000000  
3fff2060:  00000000 00000000 402175f8 3fff1090  
3fff2070:  3fffdad0 00000000 3fff1074 3fff107c  
3fff2080:  3fffdad0 00000000 3fff1074 4020de52  
3fff2090:  3fffdad0 00000000 3fff1074 40217644  
3fff20a0:  feefeffe feefeffe 3fff1090 40100718  
<<<stack<<<

Greetings,

      Heiko.

arendst, 2017-01-25:
I guess you also run Alexa Hue or Wemo?

altelch, 2017-01-25:
Yes, currently HUE an RGB as I'm working on the API. Might be related to that, too. I'll try this.

arendst, 2017-01-25:
Do you use an external power supply for the WS2812 led string? How long is your led string if powered from Sonoff? What are your settings for width, speed, wakeup, pixels?

altelch, 2017-01-25:
No, I have an USB-Hub with 3A high power port and the strip is connected to the 5V pin of the D1 mini. The RGB values are 101010 when testing, so I have a very low power usage for the strip. The strip is 1m and has 30 LEDs what should draw a maximum of 1.8A.  Width = 30, speed=5, wakeup = false pixels=30. Just recompiled without RGB support to see if the uC crashes in HUE-Mode, too.

arendst, 2017-01-25:
I suppose you're on version 3.2.10 too?

altelch, 2017-01-25:
Yes, I'm on 3.2.10.

altelch, 2017-01-25:
Ok, hammering 3.2.10 HUE API for one hour and ws2812 disabled didn't end in a crash. Seems to be LED related, will recompile again.

altelch, 2017-01-25:
Hi, I will close this. Hammering the HUE API and playing with MQTT LED commands didn't give me a single crash in 45 minutes. Must be related to my setup. Here @home everything works as expected. Sorry to bother you,

               Heiko.

closed 2017-01-25 by altelch