arendst/Sonoff-MQTT-OTA-Arduino #282: sonoff LED in next_gen

marcelveldt, 2017-01-26:
Hi again,

Sorry to spam you with yet another question.
I'm currently trying out your next_gen version of the firmware and I really like it.
So much easier to just flash the same firmware to all units and configure them through the web interface, great work!

I have a sonoff LED device and saw in the realm that it should be supported so I flashed the next_gen firmware onto the device. So far looking good but I can't figure out how to:

- set the brightness (mitt command topic and state topic)
- set the color temperature, as the soon LED has both cold and warm leds

Should this already be supported or not yet ?

arendst, 2017-01-26:
Marcel, no this is not supported yet as I'm still waiting for my Sonoff Led to arrive and start testing what I wrote ;-)

marcelveldt, 2017-01-26:
Thanks for the super fast answer. You live in the NL, right ?
If you live nearby den Bosch you can borrow mine if you want.

BTW: If you want I can share my own (very simple) code for controlling the unit with MQTT, maybe it will help you to better understand the unit.
Atm I have written my own sketches to control the sonoff units in my house but I do not have OTA and webconfig so I decided to switch to your firmware and probably help you out where I can