arendst/Sonoff-MQTT-OTA-Arduino #285: APP_POWERON_STATE is ignored

mrjemson, 2017-01-27:
APP_POWERON_STATE is being ignored (overridden) when resetting the a Sonoff TH10.
Have tried setting this to APP_POWERON_STATE 3 and 0 and in both cases the relay is engaged at power on no matter the previous state.

Please see below extract from Console at software reboot with device previously on and APP_POWERON_STATE set to 0.

`00:00:08 MQTT: Connected
00:00:08 MQTT: tele/sonoff1/LWT = Online (retained)
00:00:08 MQTT: stat/sonoff1/RESULT = {"POWER":"ON"}
00:00:08 MQTT: stat/sonoff1/POWER = ON
00:00:08 MQTT: stat/sonoff1/RESULT = {"POWER":"OFF"}
00:00:08 MQTT: stat/sonoff1/POWER = OFF
00:00:08 MQTT: stat/sonoff1/RESULT = {"POWER":"ON"}
00:00:08 MQTT: stat/sonoff1/POWER = ON
00:00:08 MQTT: stat/sonoff1/RESULT = {"Commands1":"Status, SaveData, SaveSate, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname"}
00:00:08 MQTT: stat/sonoff1/RESULT = {"Commands2":"MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, SwitchTopic, SwitchRetain, PowerRetain, MqttUnits, MessageFormat, GroupTopic, Timezone, LedState, LedPower, TelePeriod"}
00:00:08 MQTT: stat/sonoff1/RESULT = {"Commands3":"Power, Light, PowerOnState, PulseTime, BlinkTime, BlinkCount, Weblog, Webserver, DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzUpdateTimer, SwitchMode"}
00:00:08 MQTT: stat/sonoff1/RESULT = {"POWER":"ON"}
00:00:08 MQTT: stat/sonoff1/POWER = ON
00:00:09 MQTT: stat/sonoff1/RESULT = {"POWER":"ON"}
00:00:09 MQTT: stat/sonoff1/POWER = ON`

The second power event is turning off the relay successfully for a fraction of a second which I believe is called by APP_POWERON_STATE, but something else seems to be turning on the relay again (several times) afterward. Have tried with both USE_WALL_SWITCH enabled and disabled with no change in relay re-engaging. With APP_POWERON_STATE set to 3, the OFF event is not present if relay was previously on.

arendst, 2017-01-27:
I can't make anything from the log above as I see no reboots at all. PowerOnState only works after a power on reboot (not restart).

APP_POWERON_STATE changes only take effect when you also change CFG_HOLDER as is shown in the header of user_config.h.  It's way better to use command PowerOnState for this.

Are you using sleep? If so, turn it off as it interferes with most options.

mrjemson, 2017-01-31:
Not using sleep or PowerRetain.

> #define APP_POWERON_STATE      0

is set in the user_config.h

CFG_HOLDER was also incremented when uploading the new firmware.
Light was off when power was reset. Came on within seconds of being powered back up.

Here is another log. Complete rather than an extract as before.
This is after a complete power off rather than reboot (they seem to act in the same way anyway)

> 00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_95EDDF, GroupTopic sonoffs) Version 3.2.10
00:00:00 Wifi: Connecting to AP1 HomeCtrl in mode 11N as sonoff-3551...
00:00:05 Wifi: Connected
00:00:05 mDNS: Initialized
00:00:05 HTTP: Webserver active on sonoff-3551.local with IP address 10.80.75.108
00:00:07 MQTT: Attempting connection...
00:00:09 mDNS: Query done with 0 mqtt services found
00:00:09 MQTT: Connected
00:00:09 MQTT: tele/sonoff/LWT = Online (retained)
00:00:09 MQTT: tele/sonoff/RESULT = {"Info1":{"AppName":"Sonoff 8266 Module", "Version":"3.2.10", "FallbackTopic":"DVES_95EDDF", "GroupTopic":"sonoffs"}}
00:00:09 MQTT: tele/sonoff/RESULT = {"Info2":{"WebserverMode":"Admin", "Hostname":"sonoff-3551", "IPaddress":"10.80.75.108"}}
00:00:09 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
00:00:09 MQTT: stat/sonoff/POWER = ON
00:00:10 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
00:00:10 MQTT: stat/sonoff/POWER = ON
00:00:17 MQTT: tele/sonoff/TELEMETRY = {"Time":"1970-01-01T00:00:17", "Uptime":0, "POWER":"ON", "Wifi":{"AP":1, "SSID":"HomeCtrl", "RSSI":100}}

bribaird, 2017-02-01:
I believe I have the same problem with my Sonoff Touch.

Below, the status of the light was OFF, At 18:37:25 we see MQTT: Attempting connection... then at 18:38:02 we see MQTT: tele/officelight/LWT = Online (retained) and then at 18:38:02 MQTT: stat/officelight/POWER = ON.

Is there a setting I need to change to prevent the light from turning on after re-connection?

18:24:41 MQTT: stat/officelight/POWER = OFF
18:25:47 MQTT: stat/officelight/POWER = OFF
18:29:33 MQTT: tele/officelight/POWER = OFF
18:29:33 MQTT: tele/officelight/UPTIME = 3
18:29:33 MQTT: tele/officelight/AP = 1
18:29:33 MQTT: tele/officelight/SSID = BTHome
18:29:33 MQTT: tele/officelight/RSSI = 100
18:29:33 MQTT: tele/officelight/TIME = 2017-02-01T18:29:2
18:34:33 MQTT: tele/officelight/POWER = OFF
18:34:33 MQTT: tele/officelight/UPTIME = 3
18:34:33 MQTT: tele/officelight/AP = 1
18:34:33 MQTT: tele/officelight/SSID = BTHome
18:34:33 MQTT: tele/officelight/RSSI = 100
18:34:33 MQTT: tele/officelight/TIME = 2017-02-01T18:34:33
18:37:25 MQTT: Attempting connection...
18:37:26 mDNS: Query done with 0 mqtt services found
18:37:33 UPnP: Multicast (re)joined
18:37:43 MQTT: Attempting connection...
18:37:44 mDNS: Query done with 0 mqtt services found
18:37:51 UPnP: Multicast (re)joined
18:38:01 MQTT: Attempting connection...
18:38:02 mDNS: Query done with 0 mqtt services found
18:38:02 MQTT: Connected
18:38:02 MQTT: tele/officelight/LWT = Online (retained)
18:38:02 MQTT: stat/officelight/POWER = ON

Many Thanks

Brian.

mrjemson, 2017-02-03:
I have figured this out after turning on Debug level 4.
Seems there was a retained message of "1" on the group topic turning on all my units.
Not sure how it got there, but must have been during some testing I did initially.
Cleared the retained message and working correctly. Odd that this wasn't identified in the console until level 4 logging though.

closed 2017-02-03 by arendst