arendst/Sonoff-MQTT-OTA-Arduino #288: bug: wemo_emulation on Sonoff Dual

thommycgn, 2017-01-27:
Hi! 

In configuration I was set friendly_name1 +2 but only friendly_name1 is recognized by Alexa. The Second Device is not shown.

altelch, 2017-01-28:
Hi,

switch Serial or Console Debug du 4 DEBUG MORE and check the log for Response1/2/3 messages and HUE Bridge Configuration messages. The code was testet with up to 4 devices and worked. You can also call http://<yoursonoffip>/api/doesntmatter/lights to see the xml returned to the Echo. There should be the 2 friendly names.

arendst, 2017-01-28:
For clarity, Wemo emulation only supports 1 device and Hue emulation supports up to 4 devices?

altelch, 2017-01-28:
Hi,

exactly. Wemo supports only one device. The HUE code is limited by 
Maxdevices. Should there ever be a Sonoff device with more channels they 
will be supported.

Greetings,

     Heiko.

closed 2017-01-30 by arendst