arendst/Sonoff-MQTT-OTA-Arduino #291: sonoff 4CH on 8285 gives an error when uploading the firmware

ozeebrown, 2017-01-30:
Hi, After a long wait, I have received the 4CH sonoff today. I have tried to flash your firmware (latest download), but it gives at error (only on 4CH device). The image of thee err is attached.

[image]

I have pressed and hold FW/IO0 on power up to get the device in to flash mode
I have used selected ...
8285 board
1M/64SPIFFS
CPU 80MHz
Speed 115200
COM4

The firmware work well for 1CH/2CH sonoff switch.
And the device is working fine out of the box.

What could I be doing wrong?
Kind Regards.

arendst, 2017-01-30:
I still did not receive mine but according to this link https://github.com/arendst/Sonoff-Tasmota/issues/5 it must be possible to upload my firmware.

ozeebrown, 2017-01-30:
Update: Solved
I have used v3.1.3
Highly likely i have had tx rx crossed. When split everything and connect everything back on v3.1.3 Seems to be uploading fine.

closed 2017-01-30 by arendst

pcmvanveen, 2017-01-30:
I recieved a board v1.1  No need for a change on this board. Everything out of the box fine.