arendst/Sonoff-MQTT-OTA-Arduino #292: Error - FRIENDLY_NAME1

ozeebrown, 2017-01-30:
Hi, I'm trying out v3.2.13

But I'm getting a error when verifying the code.

C:\Users\GuyPERERA\Documents\Arduino\Sonoff-MQTT-OTA-Arduino-master\sonoff\sonoff.ino: In function 'void CFG_DefaultSet()':
sonoff:463: error: 'FRIENDLY_NAME1' was not declared in this scope

Full error is on attached note file.

Is there a fix to this error? Thanks.

[error - FRIENDLY_NAME1.txt](https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/files/738770/error.-.FRIENDLY_NAME1.txt)

arendst, 2017-01-30:
You're probably mixing up version 3.2.x and 3.9.x.

Although they use the same folder ```sonoff``` they are totally different and need a different host folder to compile.

ozeebrown, 2017-01-30:
You are absolutely correct. A restart fixed that problem too.
I have uploaded the firmware, but it shows the same problem of v3.9.9
I find USB power is not enough to power all 4 relays.
I have used a separate power supply as well.

But, Only the first relay is working.
As I mentioned in my other post, all 4 relays operates on buttons (after flash) on v3.1.3

Thank you for your quick replies.

arendst, 2017-01-30:
When using the 3.9.9 version did you select the correct Module using Configure Module and selected option 7?

ozeebrown, 2017-01-30:
Sorry, where can I find "Configure Module and selected option 7"?

arendst, 2017-01-30:
Only version 3.9.9 allows for online module selection.

Go to the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for background information.

Execute command ```module 7``` and let your system restart. It should now show 4 power status. Look at your SOnoff 4CH webpage for more information.

ozeebrown, 2017-01-31:
Thank you for the information.

closed 2017-01-31 by arendst