arendst/Sonoff-MQTT-OTA-Arduino #294: Compile error

Webflue, 2017-01-30:
Hi.  Been trying for many hours now to get this working.  Followed the WIKI and solved a fee but I can't get by the following error.

C:\Users\Stuart\Documents\Arduino\sonoff\sonoff.ino: In function 'void setup()':

sonoff:2766: error: 'class HardwareSerial' has no member named 'baudRate'

   if (Serial.baudRate() != Baudrate) {

              ^

exit status 1
'class HardwareSerial' has no member named 'baudRate'

I have check the ESP version and it's the latest and read the similar issue another user encountered.  I  guess that I'm missing the obvious so  any help will be greatly appreciated.

Full compile below

Arduino: 1.8.1 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Stuart\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Stuart\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Stuart\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=1M64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10801 -build-path C:\Users\Stuart\AppData\Local\Temp\arduino_build_188700 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.8 -prefs=runtime.tools.mkspiffs.path=C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -verbose C:\Users\Stuart\Documents\Arduino\sonoff\sonoff.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Stuart\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Stuart\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Stuart\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=1M64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10801 -build-path C:\Users\Stuart\AppData\Local\Temp\arduino_build_188700 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.8 -prefs=runtime.tools.mkspiffs.path=C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -verbose C:\Users\Stuart\Documents\Arduino\sonoff\sonoff.ino
Using board 'generic' from platform in folder: C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2
Using core 'esp8266' from platform in folder: C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2
Detecting libraries used...
"C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2/tools/sdk/include" "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2/tools/sdk/lwip/include" "-IC:\Users\Stuart\AppData\Local\Temp\arduino_build_188700/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10801 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266" "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\variants\generic" "C:\Users\Stuart\AppData\Local\Temp\arduino_build_188700\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2/tools/sdk/include" "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2/tools/sdk/lwip/include" "-IC:\Users\Stuart\AppData\Local\Temp\arduino_build_188700/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10801 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266" "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\variants\generic" "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\Ticker" "C:\Users\Stuart\AppData\Local\Temp\arduino_build_188700\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2/tools/sdk/include" "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2/tools/sdk/lwip/include" "-IC:\Users\Stuart\AppData\Local\Temp\arduino_build_188700/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10801 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266" "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\variants\generic" "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\Ticker" "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src" "C:\Users\Stuart\AppData\Local\Temp\arduino_build_188700\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2/tools/sdk/include" "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2/tools/sdk/lwip/include" "-IC:\Users\Stuart\AppData\Local\Temp\arduino_build_188700/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10801 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266" "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\variants\generic" "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\Ticker" "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src" "-IC:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266HTTPClient\src" "C:\Users\Stuart\AppData\Local\Temp\arduino_build_188700\sketch\sonoff.ino.cpp" -o "nul"
[... cut ...]
Using library ESP8266WebServer at version 1.0 in folder: C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WebServer 
Using library DNSServer at version 1.1.0 in folder: C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\DNSServer 
Using library ESP8266mDNS in folder: C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266mDNS (legacy)
exit status 1
'class HardwareSerial' has no member named 'baudRate'

Thanks for your anticipated help,

Stuart

arendst, 2017-01-30:
Sometime the latest is just not good enough....

I see you are using 2.3.0-rc2. Just try 2.3.0 and you'll be fine.

Webflue, 2017-01-31:
Hi,  Found out I even more issues I knew I had :-)

Now when I compile

Now have 2.3.0-rc1 and not rc-2 or am I missing something?  Sorry if I'm being a right dumb noob :-)

Cheers,

Stuart

Arduino: 1.8.1 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"

Build options changed, rebuilding all
In file included from C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\libraries\ESP8266WebServer\src/ESP8266WebServer.h:55:0,

                 from C:\Users\Stuart\Documents\Arduino\sonoff\sonoff.ino:144:

C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\libraries\ESP8266WebServer\src/detail/RequestHandler.h:7:18: warning: unused parameter 'method' [-Wunused-parameter]

     virtual bool canHandle(HTTPMethod method, String uri) { return false; }

                  ^

C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\libraries\ESP8266WebServer\src/detail/RequestHandler.h:7:18: warning: unused parameter 'uri' [-Wunused-parameter]

C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\libraries\ESP8266WebServer\src/detail/RequestHandler.h:8:18: warning: unused parameter 'uri' [-Wunused-parameter]

     virtual bool canUpload(String uri) { return false; }

                  ^

C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\libraries\ESP8266WebServer\src/detail/RequestHandler.h:9:18: warning: unused parameter 'server' [-Wunused-parameter]

     virtual bool handle(ESP8266WebServer& server, HTTPMethod requestMethod, String requestUri) { return false; }

                  ^

C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\libraries\ESP8266WebServer\src/detail/RequestHandler.h:9:18: warning: unused parameter 'requestMethod' [-Wunused-parameter]

C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\libraries\ESP8266WebServer\src/detail/RequestHandler.h:9:18: warning: unused parameter 'requestUri' [-Wunused-parameter]

C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\libraries\ESP8266WebServer\src/detail/RequestHandler.h:10:18: warning: unused parameter 'server' [-Wunused-parameter]

     virtual void upload(ESP8266WebServer& server, String requestUri, HTTPUpload& upload) {}

                  ^

C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\libraries\ESP8266WebServer\src/detail/RequestHandler.h:10:18: warning: unused parameter 'requestUri' [-Wunused-parameter]

C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\libraries\ESP8266WebServer\src/detail/RequestHandler.h:10:18: warning: unused parameter 'upload' [-Wunused-parameter]

In file included from C:\Users\Stuart\Documents\Arduino\sonoff\sonoff.ino:148:0:

C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\libraries\ESP8266mDNS/ESP8266mDNS.h:66:8: warning: unused parameter 'ip' [-Wunused-parameter]

   bool begin(const char* hostName, IPAddress ip, uint32_t ttl=120){

        ^

C:\Users\Stuart\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc1\libraries\ESP8266mDNS/ESP8266mDNS.h:66:8: warning: unused parameter 'ttl' [-Wunused-parameter]

C:\Users\Stuart\Documents\Arduino\sonoff\sonoff.ino: In function 'void publish_status(uint8_t)':

C:\Users\Stuart\Documents\Arduino\sonoff\sonoff.ino:2063:38: warning: unused variable 'stemp1' [-Wunused-variable]

   char stopic[TOPSZ], svalue[MESSZ], stemp1[TOPSZ], stemp2[10], stemp3[10];

                                      ^

C:\Users\Stuart\Documents\Arduino\sonoff\sonoff.ino:2063:53: warning: unused variable 'stemp2' [-Wunused-variable]
[... 60 lines cut ...]

                         ^

C:\Users\Stuart\Documents\Arduino\sonoff\sonoff.ino: In function 'void setup()':

sonoff:2768: error: 'class HardwareSerial' has no member named 'baudRate'

   if (Serial.baudRate() != Baudrate) {

              ^

C:\Users\Stuart\Documents\Arduino\sonoff\webserver.ino: In function 'void handleInfo()':

webserver:1191: error: 'class EspClass' has no member named 'getCoreVersion'

   page += F("<tr><td><b>Core/SDK version</b></td><td>"); page += ESP.getCoreVersion(); page += F("/"); page += String(ESP.getSdkVersion()); page += F("</td></tr>");

                                                                      ^

exit status 1
'class HardwareSerial' has no member named 'baudRate'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

arendst, 2017-02-01:
Yes you're dumb.

Sometime the latest is just not good enough....

I see you are using 2.3.0-rc2. Just try 2.3.0 and you'll be fine.

Webflue, 2017-02-03:
OK Ta - I didn't have the 2.3.0 so wrongly assumed 2.3.0-rc1 was what you meant :-(   Just putting hat with D on and going to stand in the corner!

closed 2017-02-03 by arendst