arendst/Sonoff-MQTT-OTA-Arduino #295: 4CH

scargill, 2017-01-31:
Hi

Just wondering if anyone had issues with the 4CH? I've used this software on many of my Sonoffs no problem - but yesterday I programmed a Sonoff 4CH v1.1 - all looked ok, I had selected MODULE SONOFF_2 - and it looked like it programmed but thereafter - nothing... no serial, no flashing lights etc... the only thing that comes to mind is I can't see any FLASH so possibly NOT an ESP8266 but the new cheap one??

jaysuk, 2017-01-31:
They use the ESP8265

scargill, 2017-01-31:
Am I right in saying that should be set up in the Arduino environment as an 8266 with 1 MEG? Didn't work...

jaysuk, 2017-01-31:
No. Should have an option for generic esp8265 with 1 meg

scargill, 2017-01-31:
It did - and that worked - thanks - http://tech.scargill.net/sonoff-4ch-4-channel-mains-control/

closed 2017-02-01 by arendst