arendst/Sonoff-MQTT-OTA-Arduino #298: Multiple relays and Alexa

sticilface, 2017-02-01:
Hi,

Very impressive lib.  I've written my own version, but nowhere near as complete as yours.  
I've implemented a wemo lib of my own to emulate it for alexa and it is work just fine.  However, like yourself I now have sockets with several relays on them (designed my own board with up to three mechanical or Solid state relays) but as far as I know alexa will only discover one device using wemo emulation.

If i understand correctly this is true for you too.

https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/blob/82253a75b096b8de24290d2036bb629546aefd6f/sonoff/webserver.ino#L1272

so it is not currently possible to use alexa turn off/on FRIENDLY_NAME1 .. 2 ...  3... 4... to turn off each one separately (which is what I want).  I tried looking into this. but bacically for wemo you' have to spawn a separate server on a different post for each instance. 

I believe that this might be possible which Hue though.  but by then i gave up. 

Its  shame as I'd really like to have multiple device support using wemo. 
Any ideas? (correct me if I'm wrong)

Also as a feature have you though about having sockets discover themselves... 
[image]
I serve up a webpage but you can switch between all the sockets easily to turn them on/off using this button in the top right.  They all discover each other using a UDP protocol i've written and it works very well.  Might be a nice feature.  

Andrew

altelch, 2017-02-02:
Hi,

simply comment WEMO Support in user_config.h and enable HUE Emulation. HUE supports as many devices as your hardware provides.

Greetings,

   Heiko.

arendst, 2017-02-03:
Use https://github.com/arendst/Sonoff-Tasmota and select desired Alexa function on the configuration web pages. No need to fiddle around with the user_config.h

sticilface, 2017-02-07:
I spent some time developing a plug in for alex on the esp8266 that would emulate a hue bridge so multiple switches could be controlled by alex within one device....  one to find that alex can only be paired with one bridge so that canned that...  I'm not using ha-bridge very successfully, so no need for native control.  I have over 23 home devices now.  until there is better support from alexa for devices....  it will never work amazingly emulating wemo and hue. wemo works fine for one socket...

closed 2017-02-07 by sticilface