arendst/Sonoff-MQTT-OTA-Arduino #299: compatibility issue iobroker

starfish1107, 2017-02-02:
iobroker features a mqtt-adapter. mqtt topics are arranged in a objects tree which can be used for further programming and graphical display.  This system only works, when the needed topics are either published or subscribed. Unfortunately Sonoff-MQTT-OTA-Arduino does not send  subscribe /cmnd/sonoff/POWER when starting up  or reconnecting and therefore there is no object in the tree for switching command to  the sonoff. I found a workaround by sending  subscribe /cmnd/sonoff/power  using mqtt spy  -  this makes iobroker writing the power object.  All other topics from arendst sketch are published during startup and thus written to the iobroker objects.  Automatic sending of the cmnd-topic together with stat and tele at start-up would make the sonoff modules available to iobroker and would be much appreciated. (maybe as an config option)

arendst, 2017-02-02:
I'll look into it if ``cmnd/sonoff/POWER`` is enough for iobroker. Doing ``cmnd/sonoff/POWER ON`` might trigger an unwanted power on.

starfish1107, 2017-02-02:
thats great, thanks - I checked it using mqtt-spy  issuing subscribe /cmnd/sonoff/POWER - that worked.
the name "sonoff" will vary depending on config of course.  I think that its fine with only the power command. other commands could be added manually as described above. I will be happy to test it as soon as it is built in.

arendst, 2017-02-04:
Solved in https://github.com/arendst/Sonoff-Tasmota version 3.9.11 just released.

starfish1107, 2017-02-04:
Checked - and it works. Thanks a lot.  Dont know if this is due to the Tasmota version, but I had to disable additional options as memory usage was up to 103% with the default config.

closed 2017-02-06 by arendst