arendst/Sonoff-MQTT-OTA-Arduino #300: Question: 3.2.x vs. 3.9.x

don-willingham, 2017-02-03:
I see the 3.9.x version has moved to https://github.com/arendst/Sonoff-Tasmota, will this version be maintained? If so, how long?

mrjemson, 2017-02-03:
I would also be interested to know the differences between the two projects.

hamwong, 2017-02-03:
me too. I see some snap shot in Tasmota, it looks more complicate than this one

arendst, 2017-02-03:
3.2.x will die out as 3.9.x provides on the fly functionality requested by many users (and me). I got tired of answering questions about changes to user_config.h that didn't do what the user expected as they didn't grasp the idea behind it (Its all in flash once configured and can be changed on the fly partly).

How long will this be maintained? How long will Donald Trump survive?

I will not add new features to this code.

tam481, 2017-02-05:
Hi Theo, would you be able to continue to provide the option to pre-configure settings such as DHT, Message Format etc in the override file?