arendst/Sonoff-MQTT-OTA-Arduino #303: no mqtt connection my sonoff TH 16

kamilo558, 2017-02-06:
Hello,
a few days before i received my first sonoff devices.

I installed MQtt server on my raspberry pi 3 using this wiki https://wiki.fhem.de/wiki/Sonoff#MQTT_Server_einrichten

[code]
# aus dem mosquitto Repo installieren:
wget http://repo.mosquitto.org/debian/mosquitto-repo.gpg.key
sudo apt-key add mosquitto-repo.gpg.key
cd /etc/apt/sources.list.d/
sudo wget http://repo.mosquitto.org/debian/mosquitto-wheezy.list
# oder für jessie
sudo wget http://repo.mosquitto.org/debian/mosquitto-jessie.list
sudo apt-get update
# mosquitto installieren, sowie client Befehl mosquito_sub (gehört nicht zum Server, wird aber weiter unten benötigt)
sudo apt-get install mosquitto mosquitto-clients

# MQTT Server Test
sudo service mosquitto status
[/code]

with the result:
[code]
pi@raspberry ~ $ sudo service mosquitto status
● mosquitto.service - LSB: mosquitto MQTT v3.1 message broker
   Loaded: loaded (/etc/init.d/mosquitto)
   Active: active (running) since Sa 2017-01-28 18:22:32 CET; 1 weeks 1 days ago
  Process: 629 ExecStart=/etc/init.d/mosquitto start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/mosquitto.service
           └─657 /usr/sbin/mosquitto -c /etc/mosquitto/mosquitto.conf

Feb 02 23:55:12 raspberrypi systemd[1]: Started LSB: mosquitto MQTT v3.1 mes....
Feb 03 18:39:37 raspberrypi systemd[1]: Started LSB: mosquitto MQTT v3.1 mes....
Feb 03 18:49:19 raspberrypi systemd[1]: Started LSB: mosquitto MQTT v3.1 mes....
Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
Hint: Some lines were ellipsized, use -l to show in full.
[/code]

Mqtt test on raspberry with ssh has worked fine with 2 ssh windows:
1. one:
[code]
mosquitto_sub -h localhost -v -t test
[/code]
2. one:
[code]
mosquitto_pub -h localhost -t test -m "Test Mosquitto"
[/code]
Result in 1. window:
[code]
pi@raspberry ~ $ mosquitto_sub -h localhost -v -t test
test Test Mosquitto
[/code]

In the webinterface of the sonoff console i get the following result:

[code]00:00:00 APP: Project sonoff (Topic sonoff, Fallback DVES_1697CD, GroupTopic sonoffs) Version 3.1.2
00:00:00 Wifi: Connecting to AP1 HSHKH (XXXXXXXXXXXXXX) in mode 11N as sonoff-6093...
00:00:07 Wifi: Connected
00:00:07 HTTP: Webserver active on sonoff-6093 with IP address 192.168.178.29
19:00:36 MQTT: Attempting connection...
19:01:04 MQTT: Attempting connection...
19:01:21 MQTT: Attempting connection...
19:01:39 MQTT: Attempting connection...
19:01:56 MQTT: Attempting connection...
19:02:13 MQTT: Attempting connection...
19:02:30 RSLT: tele/sonoff/UPTIME = 1
19:02:30 MQTT: Attempting connection...
19:02:48 MQTT: Attempting connection...
19:03:05 MQTT: Attempting connection...
19:03:22 MQTT: Attempting connection...
19:03:39 MQTT: Attempting connection...
19:03:57 MQTT: Attempting connection...
19:04:14 MQTT: Attempting connection...
19:04:31 MQTT: Attempting connection...
19:04:41 RSLT: stat/sonoff/POWER = ON
19:04:48 MQTT: Attempting connection...
19:05:05 MQTT: Attempting connection...
19:05:23 MQTT: Attempting connection...
19:05:40 MQTT: Attempting connection...[/code]

In arendst software i have made the following changes:
[code]#define STA_SSID               "SSID"      // Wifi SSID at home
#define STA_PASS               "password"  // Wifi password at home
#define WIFI_HOSTNAME          "%s-%04d"         // Expands to <MQTT_TOPIC>-<last 4 decimal chars of MAC address>
#define WIFI_CONFIG_TOOL       WIFI_WPSCONFIG    // Default tool if wifi fails to connect (WIFI_SMARTCONFIG, WIFI_MANAGER or WIFI_WPSCONFIG)[/code]

[code]// Syslog
#define SYS_LOG_HOST           "Ip-Adresse-Server"[/code]
[code]// MQTT
#define MQTT_HOST              "Ip-Adresse-Server"
#define MQTT_PORT              1883[/code]
"Ip-Adresse-Server" is  the ip adress of my raspberry pi. This is the ip i use for ssh to communicate with my raspberry pi.

My sonoff still has no connection to mqtt on my raspberry.

Hope for your help.

Best regards,
Kamil

arendst, 2017-02-06:
Try to install the latest version first as 3.1.2 is an old one. Download from https://github.com/arendst/Sonoff-Tasmota

Enable more logging with option 3 (``weblog 3``) and see what MQTT has to tell.

It might be one of the codes below:
```
#define MQTT_CONNECTION_TIMEOUT     -4
-4: when connecting,
    try setting the version in PubSubClient.h to the older version 3_1
    // MQTT_VERSION : Pick the version 
    #define MQTT_VERSION MQTT_VERSION_3_1  <<- uncomment this line
    Mosquitto 0.15-2 apparently does not support the newer version 3_1_1.
#define MQTT_CONNECTION_LOST        -3
#define MQTT_CONNECT_FAILED         -2
-2: means it's failing to create a network connection to the broker.
    Have you checked one of the basic http examples to verify you have the network configuration and setup correct?
#define MQTT_DISCONNECTED           -1
#define MQTT_CONNECTED               0
#define MQTT_CONNECT_BAD_PROTOCOL    1
#define MQTT_CONNECT_BAD_CLIENT_ID   2
#define MQTT_CONNECT_UNAVAILABLE     3
#define MQTT_CONNECT_BAD_CREDENTIALS 4
#define MQTT_CONNECT_UNAUTHORIZED    5
 5: I found out that the error code 5 was from cloudmqtt " not authorised".
    I had setup a new user "ESP8266Client" in cloudmqtt with a blank password, I'm guessing this isn't the way to do it.
```

kamilo558, 2017-02-06:
Can you tell me which file i have to use for an update over wifi?

kamilo558, 2017-02-06:
Ok. I installed the newest version.

now i receiver:
```

22:55:15 MQTT: Attempting connection...
22:55:16 mDNS: Query done with 0 mqtt services found
22:55:24 MQTT: CONNECT FAILED, rc -2. Retry in 10 seconds
22:55:35 MQTT: Attempting connection...
22:55:36 mDNS: Query done with 0 mqtt services found
22:55:45 MQTT: CONNECT FAILED, rc -2. Retry in 10 seconds
```

arendst, 2017-02-06:
And see above what -2 means: network problems.

rlust, 2017-02-08:
I am getting the same error message when connecting to MQTT, all was fine prior to the upgrade.

mortenmathiasen, 2017-06-20:
The connection to RPI mosquitto works fine for me using commit b88ec7f5a99d4bb6b578de9f59631c922e772030 of june 19, 2017.