arendst/Sonoff-MQTT-OTA-Arduino #305: problems compiling - 'won't fit'

btsimonh, 2017-02-15:
Hi, trying to look into why Slampher is half on all the time, but having an issue compiling the 3.9 code:

c:/data/home/arduino-1.8.0/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Simon\AppData\Local\Temp\arduino_build_942246/sonoff.ino.elf section `.text' will not fit in region `iram1_0_seg'

Any clues?

s

btsimonh, 2017-02-15:
If I define 'BE_MINIMAL', then it builds.
I'll disable some of the bits I don't need to see if I can fix my Slampher, but if I'm missing something, let me know

arendst, 2017-02-16:
Change compile options to disable debugging:

option Tools > Debug Port = disabled
option Tools > Debug Level = none

btsimonh, 2017-02-16:
those options are already set :(
I've got a compile with:
in sonoff.ino
//#define USE_DS18B20
//#define USE_DHT
in user_config:
//#define USE_DOMOTICZ 
//#define USE_DISCOVERY 
//#define USE_I2C
//#define USE_IR_REMOTE
//#define USE_WS2812

the DS18b20 was the key - as I could not disable this from user_config.

But there must be some other arduino settings I guess.  
Could you post your <user>\appdata\local\arduino1.5\preferences.txt for comparison?

Maybe we should consider a two stage upgrade process - first load a small firmware with OTA, which then bootstraps the load of the larger real firmware - you could achieve maybe a firmware size of 75% of the flash size?  Somebody has probably already done it :).....

arendst, 2017-02-16:
Using the default options in user_config.h should work just fine. Plenty of room left. No need to disable anything.

Are you sure you have set up your sources correctly? You seem to use code from https://github.com/arendst/Sonoff-Tasmota and report issues to https://github.com/arendst/Sonoff-MQTT-OTA-Arduino which makes me think you just copied the source from Tasmota over the old version which might leave files not to be used anymore.

btsimonh, 2017-02-16:
:) no, fresh checkout yesterday of tasmota.  git shows nothing modified apart from my commenting of defines.  Maybe it's library versions?
The result of the above is 456,432 bytes.
But just re-confirmed that uncommenting USE_DHT and USE_DS18B20 leads to the compilation error.

I do notice some compiler option differences in 'platform.txt' in the repo vs what's in my arduino folder.  but if I use yours, I had to fix up a bit, and the result so far is that I lose many string functions....
Mine was native 2.3.0

btsimonh, 2017-02-16:
:) playing with it I've totally stuffed my environment, even though I kept a backup.  I guess re-install from scratch is a good option now; will let you know.

btsimonh, 2017-02-16:
ok, it compiles with all options turned on in PlatformIO, to 469,136 bytes (I included DS18x20 - as I use that on one of my devices for 5 temp sensors).  Not sure I'll go back to Arduino IDE right now....

closed 2017-03-05 by arendst