arendst/Sonoff-MQTT-OTA-Arduino #307: How to notify button pressed

carmeloMagna, 2017-02-19:
Hi everybody,

thank you for your very nice code and work on SonOff. I fint it very easy and intuitive to use.

I'm writing a very basic heater control (a thermostat) switch, based on sonoff and a remote temperature reader.

What  I need is how to  get notified when the user manually switch the sonoff, in order to block the auto switch manage and report the system to be manually operated.

Somebody can help me? Thank you

glogiotatidis, 2017-02-20:
I suggest that you use MQTT. You'll need
 - an MQTT server private (suggested) or public
 - configure your sonoff with the server and then the switch will `publish` an event to the server every time the button is pressed
 - From the controller use an MQTT client to `subscribe` to the events published by the sonoff and build your logic (e.g. block the auto switch manage).

carmeloMagna, 2017-02-20:
Hi @glogiotatidis,

first of all, thank you for your comment. 
I already use MQTT, sorry if I did not mension it in my question. 
I subscribed to `stat/sonoff/#` but I cannot receive  a button event, but only `/POWER ON/OFF`. 

Do you know which topic I have to subscribe to access button event? 

Thank you.

Carmelo

glogiotatidis, 2017-02-20:
I guess you should say that if received status != stored status then it's a button event that happened. I don't have more details on how to trigger specifics mqtt messages on button events.

carmeloMagna, 2017-02-20:
Yes, what I have done is to split into two topic the auto command and the event itself.
So, if my automatic daemon publish a power on, it will do that on a different topic (i.e `encyclopedia`). Than, one other daemon take trace of that topic and the `sonoff` topic. In that way, if a direct status change is commanded and published in `sonoff` topic, the second daemon can understand it there was a button pressing event and act the poperly logic to stop the first daemon.
It's feasible, but it is very boring to program. What I expected from a so nice repository is that the button pressing publish in a `my_device/BUTTON` topic.

Since I read in documentation that the `retain` flag could be activated, I'm asking also to someone if by using the `retain` flag a button pressing event can be observed and easily used. I tried to do by myself but nothing, I can't. 

Thank you.

Rudoxxxx, 2017-02-27:
Hello! About pressed PUSHBUTTONS Ihave little problem.
In my home I have big HALL with 5x PUSH button switch

I want to use GPIO14 with a 5 PUSH button switch.
I install last version ( Sonoff and ElectroDragon by Theo Arends VERSION // 3.9.21) off programm, change to SWITCH_MODE PUSHBUTTON, in 2 places, but not help.

    Sonoff.ino Line #87:
    #define SWITCH_MODE PUSHBUTTON // TOGGLE, FOLLOW or FOLLOW_INV (the wall switch state)

    user_config.h line #113
    #define SWITCH_MODE PUSHBUTTON // [SwitchMode] TOGGLE, FOLLOW, FOLLOW_INV, PUSHBUTTON or PUSHBUTTON_INV (the wall switch state)

Thanks for help and Thanks for this project.