arendst/Sonoff-MQTT-OTA-Arduino #314: ITEAD S20 socket switches itself off

gadjet, 2017-04-22:
I just used a reprogrammed ITEAD s20 socket to control the power to my PC and it switched itself off unexpectedly.
Not sure what happened but looking at the log it seem to reconnect to the network and then power off, I then plugged it in another socket and switched it on then when it switched off later I took a screen grab of the console log.
[image]
In the log it looks like it connected and then switched off and back on (previously it just switched off)

Anyone any idea what's happening?

gadjet, 2017-04-23:
Just some more info, it looks like it tends to turn itself off and back on roughly every minute ??
[image]

sebastianlorenzen, 2017-04-23:
I'm having similar issues. Every time connect is lost to the Mqtt broker the s20 or basic is witching off an on or on and off if the device was off at this time. Connection lost could be no wifi or just no connection via the network to the broker.
It happened 5 times in the last 30 mins. This corresponds to issues in my Dlan (power lan) connection at the same time.

gadjet, 2017-04-24:
Yes it does look like it happens whenever the MQTT signs in.
Does anyone know is the device reconnecting to the wifi then MQTT or just connecting to the MQTT server?

sebastianlorenzen, 2017-04-24:
In my case wifi is connected and connection to Mqtt is lost and reconnected.
I don't use username and password for MQTT.

sebastianlorenzen, 2017-04-24:
so, I have played a bit more. I restartet my mqtt host and could repro when connection was established again.
log from a S20
`21:17:20 RSLT: DataCb Topic s20_01, Group 0, Index 1, Type POWER, Data ON (ON)
21:17:20 MQTT: stat/s20_01/RESULT = {"POWER":"ON"}
21:17:20 MQTT: stat/s20_01/POWER = ON (retained)
21:17:20 Config: Saved configuration (1364 bytes) to flash at F7 and count 594
21:18:35 MQTT: Attempting connection...
21:18:36 mDNS: Query done with 0 mqtt services found
21:18:36 MQTT: Connect FAILED to 192.168.x.x:1883, rc -2. Retry in 10 seconds
21:18:48 MQTT: Attempting connection...
21:18:49 mDNS: Query done with 0 mqtt services found
21:18:55 MQTT: Connect FAILED to 192.168.x.x:1883, rc -2. Retry in 10 seconds
-- snip ---
21:20:39 MQTT: Connected
21:20:39 MQTT: tele/s20_01/LWT = Online (retained)
21:20:39 MQTT: cmnd/s20_01/POWER = 
21:20:39 RSLT: DataCb Topic s20_01, Group 0, Index 1, Type POWER, Data TOGGLE (TOGGLE)
21:20:39 MQTT: stat/s20_01/RESULT = {"POWER":"OFF"}
21:20:39 MQTT: stat/s20_01/POWER = OFF (retained)`

the same from a basic:
`21:18:04 RSLT: DataCb Topic basic_02, Group 0, Index 1, Type POWER, Data TOGGLE (TOGGLE)
21:18:04 MQTT: stat/basic_02/RESULT = {"POWER":"ON"}
21:18:04 MQTT: stat/basic_02/POWER = ON (retained)
21:18:05 Config: Saved configuration (1364 bytes) to flash at F7 and count 466
21:18:36 MQTT: Attempting connection...
21:18:37 mDNS: Query done with 0 mqtt services found
21:18:37 MQTT: Connect FAILED to 192.168.x.x:1883, rc -2. Retry in 10 seconds
21:18:48 MQTT: Attempting connection...
21:18:49 mDNS: Query done with 0 mqtt services found
21:18:55 MQTT: Connect FAILED to 192.168.x.x:1883, rc -2. Retry in 10 seconds
-- snip --
21:20:39 MQTT: Connected
21:20:39 MQTT: tele/basic_02/LWT = Online (retained)
21:20:39 MQTT: cmnd/basic_02/POWER = 
21:20:39 RSLT: DataCb Topic basic_02, Group 0, Index 1, Type POWER, Data TOGGLE (TOGGLE)
21:20:39 MQTT: stat/basic_02/RESULT = {"POWER":"OFF"}
21:20:39 MQTT: stat/basic_02/POWER = OFF (retained)
21:20:39 RSLT: DataCb Topic basic_02, Group 0, Index 1, Type POWER, Data TOGGLE (TOGGLE)
21:20:39 MQTT: stat/basic_02/RESULT = {"POWER":"ON"}
21:20:39 MQTT: stat/basic_02/POWER = ON (retained)`

gadjet, 2017-04-24:
does the powerretain option make any difference, mine was set to off and I've changed it to on.

gadjet, 2017-04-24:
nope, the setting made no difference.
I noticed that this SW has come to the "end of life" and there is a new GIT repository with a newer version on, maybe this will have been fixed in this version, I'm going to look at it.

sebastianlorenzen, 2017-04-24:
oops, I'm already using the new tasmota version. So bad news. Missed that this is opened on the old version.

gadjet, 2017-04-25:
AH! OK then I will have to try to use other software then, If I can't fix this then It is not usable.

sebastianlorenzen, 2017-04-25:
I will open an issue for the tasmota firmware. Keep fingers crossed, that it gets more traction.

gadjet, 2017-04-25:
Hmmm! it looks like the socket is receiving an off command from the MQTT server if my interpretation of the log is correct ?
```
20:06:35 MQTT: Connected
20:06:35 MQTT: Receive topic cmnd/sonoff/power, data size 3, data off
20:06:35 MQTT: DataCb Topic sonoff, Group 0, Index 1, Type POWER, Data off (OFF)
20:06:35 MQTT: stat/sonoff/POWER = OFF (retained)
20:06:35 MQTT: Receive topic cmnd/sonoff/power/, data size 5, data Start
20:06:35 MQTT: DataCb Topic sonoff, Group 0, Index 1, Type POWER, Data Start (START)
20:06:35 MQTT: stat/sonoff/POWER = ON (retained)
20:06:36 Wifi: Checking connection...
20:06:36 Wifi: Connected
20:06:36 UPnP: Multicast (re)joined
```

gadjet, 2017-04-25:
Oops didn't mean to close it!

sebastianlorenzen, 2017-04-25:
lets keep the discussion in this [Issue](https://github.com/arendst/Sonoff-Tasmota/issues/384) as this firmware version is end of live anyways. I don't use it anymore on all my devices.

gadjet, 2017-04-25:
I think I might of found the problem, it looks like I have a node red function block that was misbehaving and writing to the MQTT server, I couldn't put my finger on exactly why but I've now re-written the function and it seems to be behaving but I will monitor it for a while before confirming it.

sebastianlorenzen, 2017-04-26:
try it out with the new firmware as well. On my end it was probably a configuration issue at the mosquitto MQTT broker. Persistence was enabled.

gadjet, 2017-04-26:
This is winding me right up, the log says that it is receiving a MQTT cmnd to switch the socket off but the log on cloudmqtt doesn't show that happening and nothing I've got is sending it .... ARGHH!!!

HobbytronicsPK, 2017-11-24:
Hi Guys, I have observed a weird thing with SONOFF S20. I was facing connectivity problems. First i though my internet had an issue, in depth analysis revealed that my internet was working smoothly. Installed my own firmware which I am using since couple of years with ESP8266 without any issue, but problem was still there.

A lot of debugging did not find any resolution. A closed observation of SONOFF S20 enclosure revealed that problem was originating from the enclosure itself. Actually, the silver colour sticker present on lower side of SONOFF S20 was causing hindrance with WiFi signals, just removed it and everything started working like a charm.

ThomasBj, 2018-03-07:
I have the same problem. Removing the sticker did not help.
The console says that MQTT is turning it off but even if a disable MQTT in the "configure other" screen
i get the same effect.