arendst/Sonoff-MQTT-OTA-Arduino #317: don't connect to homebridge

ta4ukoma, 2017-06-20:
Hi! First thx for you job!
I have some problem with sonoff and hombridge:
00:00:00 APP: Project sonoff ta4ukoma-sonoff (Topic ta4ukoma-sonoff, Fallback DVES_9706EE, GroupTopic sonoffs) Version 5.2.0
00:00:00 Wifi: Connecting to AP1 ta4ukoma-test in mode 11N as ta4ukoma-sonoff...
00:00:06 Wifi: Connected
00:00:06 mDNS: Initialized
00:00:06 HTTP: Webserver active on ta4ukoma-sonoff.local with IP address 192.168.1.174
00:00:06 UPnP: Multicast (re)joined
00:00:07 UPnP: Multicast (re)joined
00:00:08 MQTT: Attempting connection...
16:09:14 mDNS: Query done with 0 mqtt services found
16:09:14 MQTT: Connect FAILED to 192.168.1.150:1883, rc -2. Retry in 10 seconds
16:09:14 UPnP: Multicast (re)joined
16:09:24 MQTT: Attempting connection...
16:09:25 mDNS: Query done with 0 mqtt services found
16:09:25 MQTT: Connect FAILED to 192.168.1.150:1883, rc -2. Retry in 10 seconds
16:09:26 UPnP: Multicast (re)joined
settings.json:
 {
        "bridge": {
                "name": "Homebridge",
                "username": "CC:22:3D:E3:CE:30",
                "port": 51826,
                "pin": "031-45-154"
        },

        "platforms": [],
        "accessories": [{
                "accessory": "mqtt-switch-tasmota",
                "switchType": "outlet",
                "name": "ta4ukoma-sonoff",
                "url": "mqtt://192.168.1.150:1883",
                "username": "admin",
                "password": "admin",
                "caption": "room",
                "topics": {
                        "statusGet": "stat/ta4ukoma-sonoff/RESULT",
                        "statusSet": "cmnd/ta4ukoma-sonoff/POWER"
                },
                "onValue": "ON",
                "offValue": "OFF"

        }]

}
Sonoff have ip - 192.168.1.174
[image]

Can you help me ?

chris-008, 2017-10-05:
did you fill in your MQTT detail son the user_config.h file while flashing your module?

chris-008, 2017-10-05:
example, see the "else" part

#ifdef USE_MQTT_TLS
  #define MQTT_HOST            "m20.cloudmqtt.com"  // [MqttHost]
  #define MQTT_FINGERPRINT     "A5 02 FF 13 99 9F 8B 39 8E F1 83 4F 11 23 65 0B 32 36 FC 07"  // [MqttFingerprint]
  #define MQTT_PORT            20123                // [MqttPort] MQTT TLS port
  #define MQTT_USER            "cloudmqttuser"      // [MqttUser] Mandatory user
  #define MQTT_PASS            "cloudmqttpassword"  // [MqttPassword] Mandatory password
#else
  #define MQTT_HOST            "192.168.1.7"          // [MqttHost]
  #define MQTT_PORT            1883              // [MqttPort] MQTT port (10123 on CloudMQTT)
  #define MQTT_USER            "pi"       // [MqttUser] Optional user
  #define MQTT_PASS            "pipassword"       // [MqttPassword] Optional password
#endif