arendst/Sonoff-MQTT-OTA-Arduino #324: How to flash Slampher version 2.0?

andreasloe, 2017-09-09:
I am unsure how to flash the new version of Slampher v2.0 2017-04-14. First, it is a ESP8285 instead of ESP8266. I read in this 
[http://homeautomation.proboards.com/thread/259/program-slampher-socket-esp8285-esp8266](blog)
that No9 instead of No20 has to be connected to GND in order to get into flashmode. I did all this but I am still stuck with the following error message
`esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 57600
	setting port from /dev/tty.usbserial to /dev/cu.SLAB_USBtoUART
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.SLAB_USBtoUART at 57600
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed`

andreasloe, 2017-09-09:
I am currently asking this in papa's blog, it seems to be an issue with getting into flashing mode:
[http://homeautomation.proboards.com/thread/259/program-slampher-socket-esp8285-esp8266?page=1&scrollTo=2600](url)

EvertDekker, 2017-09-29:
For version 2.0 you need to move R21 to R3: http://evertdekker.com/?p=954

andreasloe, 2017-09-29:
I am sorry that I did not write the solution here. It can be found if you follow the link above. You have to connect the left pad of R9 to GND in order to get into flash mode. Simply read until the end, the issue can be closed.

closed 2017-09-29 by andreasloe