arendst/Sonoff-MQTT-OTA-Arduino #334: Screwed up web GUI by choosing wrong module

Ontrackx, 2018-05-29:
I have a 4ch module
I screwed around in config and chose wrong module
The 4chan is talking on my mqtt bus
and i can reach it from http
But how to issue change to 4channel mode via http?

like, I just get command uknown
http://sonoff/cm?user=admin&password=******&cmnd=modules

arendst, 2018-05-29:
BTW this repository is end-of-life

Ontrackx, 2018-05-29:
I am bit of a neewb to mqtt
Http only shows
яяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяя  л
If i try with http
http://sonoff/cm?user=admin&password=****&cmnd=Modules% i get

{"Command":"Unknown"}

I think i just write the http string wrong

Ontrackx, 2018-05-29:
Sorry i am in the wrong section i seee....

arendst, 2018-05-29:
When it shows that it probably also doesn't handle your http modules request ...

Ontrackx, 2018-05-29:
what will the command be to try change module to 4ch sonoff with mqtt

i can try that
Sorry tried to read how to but cant seem to find the correct syntax

Ontrackx, 2018-05-29:
You don't possible have time to look at it via teamviewer? I know it is a stupid question
But if this happens to others like me :) maybe it can be diagnosed 
Btw i am on the most recent binary release and it was working perfectlu util I had to test what happend if i change to wrong module:
My mqtt server:
: Sending PUBLISH to mqtt-spy (d0, q0, r0, m0, 'cmnd/sonoff/reset', ... (3 bytes))
1527611274: Sending PUBLISH to Sonoff4ch (d0, q0, r0, m0, 'cmnd/sonoff/reset', ... (3 bytes))
1527611274: Received PUBLISH from Sonoff4ch (d0, q0, r0, m0, 'stat/sonoff/RESULT', ... (21 bytes))
1527611274: Sending PUBLISH to mqtt-spy (d0, q0, r0, m0, 'stat/sonoff/RESULT', ... (21 bytes))
1527611281: Received PUBLISH from mqtt-spy (d0, q0, r0, m0, 'cmd/sonoff/reset', ... (3 bytes))
1527611281: Sending PUBLISH to mqtt-spy (d0, q0, r0, m0, 'cmd/sonoff/reset', ... (3 bytes))

Ontrackx, 2018-05-29:
Thankyou :)