arendst/Sonoff-MQTT-OTA-Arduino #335: sonoff basic DHT22/AM2301 TEMP/Humidity wrong value displayed

blueboi69, 2018-06-06:
Hi!
First of all, very big thanks for this firmware :)

I have an issue with sonoff basic humidity readings. I use firmware 5.14.0. I realised that a very high humidity is displayed. Between 90-100% This is impossible! I tried with multiple DHT22 sensors, multiple sonoffs, all the same.
I checked the console in debug mode and converted the sensor values from HEX to DEC. The TEMP reading is the same as displayed on the home page, but the HUMIDITY it not! The HEX value is correct, the displayed value is not.

For example, on the console I read this:
DHT: Received 03, 9A, 00, F7, 94 =? 94

where 9A is the Humidity 155, which should be 15.5% (still I think its low, but ok) 
F7 is TEMP, which is 247 which means 24.7°C thaths correct
Meanwhile on the home page I see:

AM2301 Temperature | 24.7°C
AM2301 Humidity | 92.2%
I also get the same wrong result through MQTT

Do you have an idea what could be wrong?

arendst, 2018-06-06:
I do not have a DHT22 on hand but two AM2301 sensors report it correctly:
```
10:11:23 DHT: Received 02, 5A, 00, E2, 3E =? 3E
10:11:23 DHT: Received 02, 6D, 00, E1, 50 =? 50
10:11:23 MQT: tele/wemos4/SENSOR = {"Time":"2018-06-06T10:11:23","AM2301-12":{"Temperature":22.6,"Humidity":60.2},"SI7021-14":{"Temperature":22.5,"Humidity":62.1},"TempUnit":"C"}
```
Both DHT22 and AM2301 use two bytes for the humidity:
So 025A = 602 = 60.2%
In your case it is 039A = 922 = 92.2% as a DHT22 uses two bytes

Only a DHT11 uses the first byte for humidity and the third byte for temperature:
```
    switch (Dht[sensor].type) {
    case GPIO_DHT11:
      h = dht_data[0];
      t = dht_data[2];
      break;
    case GPIO_DHT22:
    case GPIO_SI7021:
      h = ((dht_data[0] << 8) | dht_data[1]) * 0.1;
      t = (((dht_data[2] & 0x7F) << 8 ) | dht_data[3]) * 0.1;
      if (dht_data[2] & 0x80) {
        t *= -1;
      }
      break;
    }
```

blueboi69, 2018-06-06:
Hmmm.. Thanks for the fast response! I have to find a calibration method as I really don't think there is more than 90% humidity all day.

Thanks, this means that I was wrong and it is displayed correctly, you can close this issue. 

Also if you have an idea on calibration, let me know.